ut-utp/core #14: Rethink the interrupt interface for all the traits

rrbutani, 2019-11-17:
### what
Ditch Interrupts on Adc, probably.

Input will also need to grow to have interrupts _and_ should probably grow a cache for the last character (effectively is the _KBDR_).

(Note: I _think_ none of the other interrupts need this since none of them have a 'payload' (i.e. data associated with an interrupt) but we should talk about this.)

But mainly, switch to using [AtomicBool](https://doc.rust-lang.org/core/sync/atomic/struct.AtomicBool.html)s instead of our weird double-indirection-with-functions scheme.

(Note: the atomics have interior mutability which is why we can do this without suffering.)

This would likely result in the traits with interrupts growing a function to check that an interrupt has occurred (if the traits now own the bit of shared state that is). Which leads to the question: How to work around the `'static` requirement and how should peripheral impls actually expose their state?

One way is for each peripheral _trait_ to come with a `pub static` for the 

### steps
 - [ ] Discuss. (This is still in the primordial stages so no real steps yet!)

### where
branch: `feat-peripheral-trait-reform`

### open questions
The above about covers it for now, I think.

Wait, here's one:
  - Are the [platform restrictions on atomics](https://doc.rust-lang.org/core/sync/atomic/index.html#portability) a concern?

rrbutani, 2019-11-17:
We could have the user of the simulator (at which point the platform is _known_) provide the shared state.

This isn't a great interface and might require us to remove the Default requirement, but it _should_ work. (if we don't want to remove the Default requirement we could add functions that let you set a reference to the state.. at which point we'd have actually come full circle to `register_interrupt`)

rrbutani, 2019-11-17:
Here's an somewhat minimal example of the last comment:

```rust
use core::marker::PhantomData;
use core::sync::atomic::{AtomicBool, Ordering};

struct Sim<'a> {
    flag1: Option<&'a AtomicBool>,
    flag2: Option<&'a AtomicBool>,
    pub perip1: Peripheral<'a>,
    pub perip2: Peripheral<'a>,
}

impl Default for Sim<'_> {
    fn default() -> Self {
        Self { flag1: None, flag2: None, perip1: Peripheral::default(), perip2: Peripheral::default() }
    }
}

impl<'a> Sim<'a> {
    fn new() -> Self {
        Self::default()
    }

    fn register_flag1(&mut self, flag: &'a AtomicBool) {
        self.flag1 = Some(flag);
        self.perip1.register_flag(flag);
    }

    fn register_flag2(&mut self, flag: &'a AtomicBool) {
        self.flag2 = Some(flag);
        self.perip2.register_flag(flag);
    }

    fn step(&self) {
        print!("(step)");

        if self.perip1.interrupt_occurred() { print!(" {{Interrupt 1!}}"); }
        if self.perip2.interrupt_occurred() { print!(" {{Interrupt 2!}}"); }

        println!("");
    }
}

struct Peripheral<'a> {
    flag: Option<&'a AtomicBool>,
    interrupts_enabled: bool,
}

impl Default for Peripheral<'_> {
    fn default() -> Self {
        Self { flag: None, interrupts_enabled: false }
    }
}

impl<'a> Peripheral<'a> {
    fn register_flag(&mut self, flag: &'a AtomicBool) {
        self.flag = Some(flag)
    }

    fn enable_interrupts(&mut self) {
        self.interrupts_enabled = true;
    }

    fn disable_interrupts(&mut self) {
        self.interrupts_enabled = false;
    }

    fn interrupt_occurred(&self) -> bool {
        match self.flag {
            Some(b) => {
                let occurred = b.load(Ordering::SeqCst);

                if self.interrupts_enabled && occurred {
                    b.store(false, Ordering::SeqCst);
                    true
                } else {
                    false
                }
            },
            None => unreachable!(),
        }
    }
}

// And for fun maybe we have this:
struct Set;
struct NotSet;
struct SimBuilder<'a, Flag1, Flag2>(Sim<'a>, PhantomData<Flag1>, PhantomData<Flag2>);

impl SimBuilder<'_, NotSet, NotSet> {
    pub fn new() -> Self { Self(Sim::new(), PhantomData, PhantomData) }
}

impl<'a, F2> SimBuilder<'a, NotSet, F2> {
    pub fn set_flag1(mut self, flag: &'a AtomicBool) -> SimBuilder<Set, F2> {
        self.0.register_flag1(flag);

        SimBuilder::<Set, F2>(self.0, PhantomData, PhantomData)
    }
}

impl<'a, F1> SimBuilder<'a, F1, NotSet> {
    pub fn set_flag2(mut self, flag: &'a AtomicBool) -> SimBuilder<F1, Set> {
        self.0.register_flag2(flag);

        SimBuilder::<F1, Set>(self.0, PhantomData, PhantomData)
    }
}

impl<'a> SimBuilder<'a, Set, Set> {
    pub fn build(self) -> Sim<'a> {
        self.0
    }
}

// We'd maybe want to bundle this with the peripheral impl; i.e. embedded peripheral
// impls expose wrapped types that have all this? Or something.
static GLOBAL_FLAG1: AtomicBool = AtomicBool::new(false);
static GLOBAL_FLAG2: AtomicBool = AtomicBool::new(false);

use rand::prelude::*;
use std::thread;
use std::time::Duration;

fn isr_function_1() {
    // Randomly trigger interrupts.
    let next = thread_rng().gen_bool(0.1);
    if next {
        GLOBAL_FLAG1.store(next, Ordering::SeqCst);
        println!("I1");
    }
}

fn isr_function_2() {
    // Randomly trigger interrupts.
    let next = thread_rng().gen_bool(0.1);
    if next {
        GLOBAL_FLAG2.store(next, Ordering::SeqCst);
        println!("I2");
    }
}

fn main() {
    let isr1 = thread::spawn(|| loop {
        isr_function_1();
        thread::sleep(Duration::from_millis(500));
    });

    let isr2 = thread::spawn(|| loop {
        isr_function_2();
        thread::sleep(Duration::from_millis(500));
    });

    let mut sim = SimBuilder::new()
        .set_flag1(&GLOBAL_FLAG1)
        .set_flag2(&GLOBAL_FLAG2)
        .build();

    sim.perip1.enable_interrupts();
    sim.perip2.enable_interrupts();

    loop {
        sim.step();
        thread::sleep(Duration::from_millis(500));

        if false {
            break;
        }
    }
}
```

gipsond, 2019-11-21:
I think we should consider somehow tying together the `Interrupt` state or `interrupts_enabled` `bool` you suggest to the `AtomicBoolean` so the compiler knows about the precondition that the `AtomicBoolean` has to be passed in before calling interrupt_occurred.

Maybe when activating interrupts, we can always require passing in a reference to the static flag you want to use?

gipsond, 2019-11-21:
Mostly finished on `feat-peripheral-shims` but how shims handle interrupts is still fuzzy

closed 2020-02-02 by rrbutani