ut-utp/core #188: RPC improvements

pranav12321, 2022-08-31:
### what
1) Checksums
2) Step function counter
3) Stress tests and error handling
4) DMA HAL traits and UART break line feature
5) Potential support for wireless communication Sub GHZ bandwidth
[[
What is this improvement about?
Here's an example: "Add doc tests for the `AddImm` constructor in `lc3-isa`"
]]

### steps
 - [ ] [[ steps ]]

[[ What steps are needed to implement this improvement? ]]

### where
branch: `imp-`

[[
This should be:
 - N/A if this improvement is not currently being worked on
 - [`imp-<improvement-name>`](//github.com/ut-utp/<repo>/tree/imp-improvement-name) if this is being worked on in _this repo_
 - [`org/repo#<branch-name>`](//github.com/org/repo/tree/branch-name) if this is being worked on in _another repo_
]]

[[
Some improvements are closely tied to code that's already in the codebase.
i.e.: "update `Instruction`'s `Display` impl to print opcodes as hex when alternate the format specifier is used" should probably link to where the current `Display` impl for `Instruction` is

For such improvements, it's helpful to link to these bits of code here; we ask that you use permalinks to code in `master` to do so.
As an example, permalinks look like this: [lc3-isa `Word` type](https://github.com/ut-utp/core/blob/4816ece0f2d47e54d989c7a5bc4da9c9f5415f74/isa/src/lib.rs#L82-L83)

See [here](https://help.github.com/en/github/managing-files-in-a-repository/getting-permanent-links-to-files) for details about how to make permalinks on GitHub.

For improvements where this isn't applicable, feel free to remove this list.
]]

relevant code:
  - [example link](https://github.com/ut-utp/core/blob/4816ece0f2d47e54d989c7a5bc4da9c9f5415f74/isa/Cargo.toml#L2)

### open questions

[[ Open questions about this improvement. ]]

rrbutani, 2022-08-31:
@pranav12321 just to be sure: by step counter you mean a message numbering scheme that lets us safely retry non-idempotent messages on error?

rrbutani, 2022-08-31:
I think this is a good list of RPC related stuff we'd want to explore in the future!

All of 1-4 seem like well-scoped self contained tasks that would be great candidates for issues/PRs of their own (that can be tracked by this issue) when work begins on these tasks.

5 does seem like it should have it's own issue though; I'm a little fuzzy on what exactly it entails but that's okay.

rrbutani, 2022-08-31:
@pranav12321 I've filled in some context on the original issue message; please feel free to edit/add to it